tonnico/Marlin4Due #3: MultiU24X24toH16 with u_long timer

Wurstnase, 2015-02-03:
Hi @nophead, hi @alexborro,

in https://github.com/MarlinFirmware/Marlin/issues/1396 you've talked about fix point notation. I guess my main-problem should be here.

In stepper.cpp my timer is long, because the Sam3x can easily handle this. In some cases the `timer` could be bigger than 24bit.

```
#define MultiU24X24toH16(intRes, longIn1, longIn2) intRes = ((uint64_t)(longIn1) * (longIn2)) >> 24
```
 and

```
block->acceleration_rate = (long)((float)block->acceleration_st * (16777216.0 / (HAL_TIMER_RATE)))
```

where HAL_TIMER_RATE is 42,000,000.

Did you think, this can be easily fixed?

Many thanks in adcance
Nico

nophead, 2015-02-03:
> in MarlinFirmware/Marlin#1396

Not sure it is broken. block->acceleration rate is 8 bits integer, 24 bits
fraction. Is that not sufficient?

If it isn't you need to change the >> 24 and the 16777216.0 to allow more
bits of integer or more bits of fraction but you will also need to change
where it is used.

Wurstnase, 2015-02-03:
Ok, maybe I understand it now. Maybe...
This could be not the problem. I think this should be a problem at lower speed (if it would be one) and not at high speed.

At the moment I search for an acceration-issue.

The acceleration works until the acceleration ends and the max speed is reached. At this point the speed jumps a bit. I can see and hear it. Any ideas where I need to look at?

nophead, 2015-02-03:
All the places it sets the timer in stepper.cpp

alexborro, 2015-02-03:
Nico, change the code to print out the OCR1A (through serial)... as the bot
speed up, it decrease, than keep constant for constant speed and then
increase as the bot slow down.
You'd be able to see any glitch.

Cheers.

Alex.

nophead, 2015-02-03:
Is is safe to call print routines from an interrupt? I would think not.

alexborro, 2015-02-03:
Yes, it is.. I'm doing a new feature and using it a lot for debugging.
Actually exactly as I told him.

Cheers.

Alex.

Wurstnase, 2015-02-03:
Sounds good for debugging. 
Do you use any special software to read the values? I think a normal monitor should not work really well?

alexborro, 2015-02-03:
Nope, just connect to the printer using pronterface.

Check the code bellow, it is placed in the beginning of the ISR:

if (current_block != NULL) {

  SERIAL_ECHO("OCR1A: ");
  SERIAL_ECHOLN(OCR1A);

  // Set directions TO DO This should be done once during init of
trapezoid. Endstops -> interrupt
  out_bits = current_block->direction_bits;
...

It will print out the OCR1A value each time the drivers are pulsed.

Cheers.

Alex.

Wurstnase, 2015-02-03:
Some hours later... 
For the Due there is no OCR1A. But I found something for anyone how search also: <a href=https://android.googlesource.com/kernel/lk/+/upstream-master/platform/sam3/drivers/tc/tc.c>tc.c</a>

```
int tc_read_rc(Tc * p_tc, uint32_t ul_channel)
{
	assert(ul_channel <
			(sizeof(p_tc->TC_CHANNEL) / sizeof(p_tc->TC_CHANNEL[0])));
	return p_tc->TC_CHANNEL[ul_channel].TC_RC;
}
```

Wurstnase, 2015-02-04:
Well, it doesn't work in the interrupt. This will slow down the complete part by factor 10.

Wurstnase, 2015-02-05:
After a half night of testing.

With a CoreXY you've got 2^16 step with sqrt(2). With a maximum of 2^16/(sqrt(2)) as quadstepping frequency the problem is gone. There could be some variable anywhere which makes the trouble.

I really think that something in the timer calculation in stepper.cpp. The Multi24x24 maybe. The complete timing goes well until the max acc_step_rate goes higher as the nominal_step_rate and acc_step_rate = nominal_step_rate.

Can you explain me the MultiU24x24toH16 at the new timer calculation with numbers?

Wurstnase, 2015-02-05:
I pictures

```
speed
|		  *****************
|	
|		
|       *				    *
|     *						  *
|   *						    *
| *							      *
_______________________________________________ -> time
```

Wurstnase, 2015-02-08:
Hi,

I'm not sure what the problem is, but to take a look deeper I made a python-script for a stepper-simulation. The script is really close to the original stepper.cpp and planner.cpp and you can see a graphic which stepper-speed you have while accerate and decelerate.

Wurstnase, 2015-02-23:
Finally i got it. 
Division is also on the Due to slow!

closed 2015-02-23 by Wurstnase