tonnico/Marlin4Due #23: Question:What configuration currently use this FW ?

luc-github, 2015-04-28:
Hi,
Currently I have a DUE/RADDS and a Davinci (Due based but different pinout)  using repetier, and for a while I wanted to jump to Marlin, but no time, and as BC fw was stuck, I give up until I saw your pull request on his github :smile: 

It looks very nice work what you did - I am curious about the configuration you have to test/use this marlin for DUE - RADDS with 128 steppers ? Just to start with a working configuration first as reference, then try the others

Wurstnase, 2015-04-28:
Actually you can find my configuration [here](https://github.com/Wurstnase/Marlin4Due/blob/debug_interrupts_1/Marlin/Configuration.h)
My setup is a CoreXY with Due, RADDS V1.5 and RAPS128. I run that with 1/128 µs.

I have only RAPS128 (1/128 stepper driver polulu-style) at the moment to test and they work perfect for me. Today I get a THB6128 for testing and next weeks some DRV also. But this needs some rework, because the next patch get some speed improvement, and stepper like THB und DRV are to slow and need some delays.

luc-github, 2015-04-28:
Ok excellent, thanks a lot, I just wanted to be sure about the reference configuration.
My printers are cartesian ones, like you the RADDS has 1/128 stepper driver and but the Davinci board has 1/16 stepper driver.
Both use currently repetier with watchdog enabled - if you need I can help porting Watchdog to Marlin because I will also work on support for the Davinci board which is a Due with special keypad/LCD

Wurstnase, 2015-04-28:
Sure. Go for it. Watchdog would be a nice feature.

luc-github, 2015-04-28:
Ok - I have forked you repo and will work on porting first to get familiar with the code and then will add the watchdog support

Thanks for your great work - it give more choice in Due world

closed 2015-04-28 by luc-github