tonnico/Marlin4Due #24: Question: ANALOG NUMBERING vs Pin Numbering

luc-github, 2015-05-01:
Hi I have a question about temperature sensor pins:

I was trying to use the ANALOG NUMBERING like in repetier FW and as mentioned in comments for pin sensor:
```#define TEMP_BED_PIN       14   // ANALOG NUMBERING```

but was not working so I used direct pin numbering (variant.cpp one) 
```#define TEMP_BED_PIN       126   // PIN NUMBERING```

and it worked, I got my sensor working and giving right temperature, so it is fine but I would like to understand.

It is normal ?

Thanks

Wurstnase, 2015-05-01:
Normally you use an analogRead for the values. And thee you use the analog numbers. Anyway, which board do you use? It's not an Arduino Due?!?

luc-github, 2015-05-01:
Haha - I wish it would be so simple ^_^
Davinci  is microcontroller board based on the Atmel SAM3X8E ARM Cortex-M3 CPU, same as arduino DUE, but the board use some of the pins differently so variant.cpp is currently modified to access some pins that are not available.
like PB24 is mapped to pin 125 used for LCD when PB24 is not available in normal DUE
PB21 is used for the Analog 14 pin 126, when PB21 is pure digital in normal DUE
PB11/PB10, pin 85 used for USB on DUE is used for the LED on DAVINCI and need to be splitted because PB10 is used for the door sensor.
So once variant.cpp is modified it can be programmed like an arduino DUE.

Back to the original question, I suspect my variant.cpp/.h missing some information for Marlin FW so I cannot use Channel 13 and Channel 14 like in repetier for temperature sensor but need to use pin number 65 and 126. I do not understand why repetier is able to handle, but I guess it does not care too much of variant.h but only on variant.cpp

My knowledge of arduino is not extensive, so I wondered why something work in one FW and not in another in same way, so I raised question to you :smile:  - anyway it is working using pin number, so it is not a big issue - just willing to understand.
Thanks for your feedback ^_^

luc-github, 2015-05-03:
I close it - as not blocking - thanks

closed 2015-05-03 by luc-github