tonnico/Marlin4Due #25: Question : Keypad

luc-github, 2015-05-01:
Sorry again I need some advice, I am not familiar with Marlin code so I request your help.
I have implemented sensors, LCD, buzzer now it is time for keypad.

DaVinci printer has a key pad and no encoder.
The key pad is actually made by 6 buttons (up, down, back, next, Ok, Home) connected to 6 different pins.
What is the best way to implement navigation with these 6 buttons ? Do you have any suggestion or part of the FW I could refer to ?

Thanks in advance,

Luc

Wurstnase, 2015-05-01:
There is already some code for a reprap keypad. You should look for that and use it for your pad.

luc-github, 2015-05-01:
I have checked quickly before, but I felt it was using one encoder for all keys when the davinci key pad is actually one pin for each button and each key is actually just a pulldown switch.

Thanks for your suggestion I will check again

luc-github, 2015-05-02:
Ok so I finally used the framework of ULTIPANEL but need to modify lcd_buttons_update() function a little to add DAVINCI_PANEL support inside
I use BTN_ENC for "OK" key, BTN_BACK for "BACK" key and BTN_EN1/BTN_EN2 for "UP" and "DOWN" key but I use the change of state to simulate a pulse and so far so good.
I can see Marlin FW is not taking lot of usage of "BACK" key so I will add it 
I will add support for the 2 last keys I have : "TOP MENU" key to got back to main screen but for "RIGHT" key I will see when I will review navigation 

I meet some compilation issue when I disabled SD Support using ULTIPANEL seems some ```#ifdef SDSUPPORT``` are missing I had to disable it as I currently have a conflict with pin 76 used for SD on DUE but on Keypad on DAVINCI
Do you want I push a PR for the 3 missing ```#ifdef SDSUPPORT``` ?

Wurstnase, 2015-05-02:
Sure, PR will help :) I can't see everything.

luc-github, 2015-05-02:
Ok will prepare this and do it tomorrow

luc-github, 2015-05-03:
PR done - I have also added a modification that is useful - allow pins_[board].h to define MYSERIAL 
and it is backward compatible as if not defined - it will define MYSERIAL as before

closed 2015-05-03 by luc-github