tonnico/Marlin4Due #29: Auto Bed level not working on delta

giutrec, 2015-05-17:
Hi, 
I've modded a ramps fd v1 (for bugfixing now is same of v2)  and i've tryed your version of Marlin for Arduino Due, it's print but not work the auto bed leveling, the printer make a probe in the same point...

leseaw, 2015-05-17:
Invert the endstops
Make m119
If endstop not open invert
X max  y max z max

Wurstnase, 2015-05-17:
When do you download it? Just fix today an issue.

giutrec, 2015-05-22:
Hi, 
Today i've tryed a new version of your Marlin4Due, now its probe in more point but in the positioning the hotend beats on the bed and fails the probe, i've attached a video :)

https://www.youtube.com/watch?v=NkgyTs0InSg

I apologize for the mess in my lab :D

Wurstnase, 2015-05-22:
I would guess that this could be also an issue with MarlinFirmware. @thinkyhead can you confirm some issues like that? I have no experience with Deltas.

thinkyhead, 2015-05-22:
@glutrec My personal apologies for your bed leveling issues. I'm not sure if these are involved in the behavior you're witnessing, but we're trying to track down a couple of issues in the Development HEAD:
- A discrepancy in leveling height on machines with servo probes (seems to happen at the end of leveling).
- The first "raise z" move before probing may fail to take place.

I have not seen an issue where the Z probe fails to deploy.

Please post your `Configuration.h` and `Configuration_adv.h` files here, in a Github Gist, or on Pastebin. If you're moving from an earlier version of Marlin to the latest, be sure to compare your configuration files to the latest configuration files, as there have been some changes. You may find it easiest to start afresh.

Once we can ensure your configuration is correct, then we can see if any of our outstanding issues are affecting you.

There are two branches posted just to test leveling and print out detailed logs: [debug_leveling](https://github.com/thinkyhead/Marlin/tree/debug_leveling) and [debug_leveling_set_z](https://github.com/thinkyhead/Marlin/tree/debug_leveling_set_z), with minor differences between them. Since you're using "Marlin4Due" I don't know if you can actually test these. If so, we would be able to look at the logs and figure out if the first raise is incorrect, or if something else is happening.

Wurstnase, 2015-05-22:
I can make these debug-branches. But sadly I'm away for a week.
When I'm back, i will start with that.

Wurstnase, 2015-05-23:
Just had the time to look that video with sound. Looks like you are loosing a lot of steps.

Can you tell me more about your setup? Stepper driver? Microsteps?
How does your probe work?

giutrec, 2015-05-23:
Ok, 
Thank's for your support :)

It states that with Arduino Mega and marlinfirmware everything works perfectly.

This is a my configuration.h and configuration_adv.h
https://gist.github.com/giutrec/52e002fa135b3c9d1588
https://gist.github.com/giutrec/16c0000f420e822cbb5c

My stepper driver is A4988 stepstick with microstepping enabled (1/16 )

@Wurstnase: Yes I'lose a step beacuse the hotend in positioning the hotend beats on the bed and the stepper loose a steps.

Now I try yours debug versions and paste you the log.

Thanks :)

Wurstnase, 2015-05-23:
> Yes I'lose a step beacuse the hotend in positioning the hotend beats on the bed and the stepper loose a steps.

OK, I don't saw this in the video. Looks like there that the hotend stops in the air.

Can you please describe exactly what happens?

giutrec, 2015-05-23:
Ok, a delta printer go home on Xmax Ymax Zmax on top of printer, now a make a video and post it for you

giutrec, 2015-05-23:
Ok this is a video 
http://youtu.be/Bf0Ji9KfJbM (is avaible in 15 mins).
But, do you think that the lack of the EEPROM will affect the problem of probe?
Because, my ramps fd not have a EEPROM....

Wurstnase, 2015-05-23:
I'm currently on an airport and can't visit your video. But I have an other question. What is the signal of your attiny? The Due can only handle 3.3V!

giutrec, 2015-05-23:
I use a logic level translator, but in future i'll modify my attiny for power on 3.3v 

Now i've tryed a little solution for the auto probe problem, run some test and report you my progress.

good flight!

Wurstnase, 2015-05-23:
> Ok, a delta printer go home on Xmax Ymax Zmax on top of printer, now a make a video and post it for you

Right. I mean the G29 part. This looks like it stops in the air at the end of your first video.

thinkyhead, 2015-05-24:
> the system works perfectly with marlinfirmware

Ok, well don't worry about the debug testing. I have others who are testing that and we are close to some resolution.

giutrec, 2015-05-29:
Hi,
Now the auto bed leveling works "fine", with a little trick... the printer make a non linear moviment when go to next point rows (concave movement, example image http://i1.wp.com/builda3dprinter.eu/wp-content/uploads/2014/05/Concave.png) and beats on the bed.
Then i have modified a #define Z_RAISE_BETWEEN_PROBINGS  from 5 to 15 and now the concave movement is over the bed and not beats, the probe works fine and i can print.

Also more pin of ramps-fd is wrong i'm correcting them (now i try to find a LCD pin on Rampsfd) when i finish make a pullrequest 

Watch the video :) https://www.youtube.com/watch?v=2MCSLfz4HgM

thinkyhead, 2015-06-15:
The curvy movements are caused by the leveling routines not using the delta kinematics, but just linearly interpolating the XYZ positions. There has been discussion here concerning how to address it, but so far no one has made the patch. And I must admit, I'm not entirely sure where to patch this to make it do proper non-curvy movements.

clefranc, 2015-10-10:
I don't have my DUE installed yet, but I've patched the curvy moves while doing G29 (8-bit Marlin). If someone is interested to port into DUE, it's there: https://github.com/clefranc/Marlin4Delta/commit/c3bfd3c5589b06474bc187600ad1b3c6e7b07d68.

I can give more information if needed.

giutrec, 2015-10-27:
ok thanks i just try to port it!