tonnico/Marlin4Due #33: Marlin not read any SD card.

giutrec, 2015-06-05:
Hi,
I've created a little pcb for connect a OLED display, one rotary encoder and an one SD card reader.
The system works quite good, buy marlin don't read any sdcard (same partition type, fat32), i think the problem is in SD libraries (for istance isn't present a SDutils library).

Wurstnase, 2015-06-05:
I don't have any issue to read a sd card. I use a RADDS with hardware spi. I think software spi is not implemented.

luc-github, 2015-06-06:
I can use SD also but using SW SPI is not working, I had to do some change to be able to use any pins others than normal ones as MOSI_PIN, MISO_PIN, SCK_PIN are defined for each board

Wurstnase, 2015-06-06:
Afaik SW spi is not implemented.

luc-github, 2015-06-06:
Well flag SOFTWARE_SPI and some code is there, but as I did not wanted to change too many files due to this SOFTWARE_SPI flag, I created another one, which allowed limited changes, https://github.com/luc-github/Marlin4Due/commit/9dd636a7cb373ff4aa375dcd69a3a2b9352ea544, so this equivalent of software SPI

giutrec, 2015-06-15:
I not use a SW SPI, the problem is show with normale SPI, i cant read any SD card but the micro SD card (using adapter) works finely.
In my display board i use directly  3.3 v for powering and signals.

I think problem is in SD utils library (is not same of marlin firmware) but in this days i havent time for search a solution....

Wurstnase, 2015-06-15:
which extension board do you use?
I use a RADDS and the Display for it. I don't have any issues with the display-SD. I only can't send files to it. Printing works.

giutrec, 2015-06-16:
I builded myself an estensione board, because for OLED display isnt exist any solution...
But, since the SD card works directly to 3.3v i have    male a solution same ad this
http://api.ning.com/files/IvqemBCauJE3Uyd42k-NHQH5vgY9BiFjYC1-06rOGN21ckAZzF2bsBNu2dJVO6JLbiOXxIsOvz-9*jI4PtMFoNmRUKopWJP9/Due_and_SD_001.png

closed 2015-07-28 by Wurstnase