tonnico/Marlin4Due #34: Watchdog Implementation

luc-github, 2015-06-15:
I did the import of watchdog code from repetier code, It seems to work but did not tested during print time yet  - https://github.com/luc-github/Marlin4Due/commit/a82336fe404ec397f4aef40c1d88129e93e9b5dc

There is also a function to test it  -  and of course need to update the Due variant.cpp to enable watchdog
if you want I can do a PR - but I am not yet familiar with marlin coding rules so you may have a review first

NB: I read future version of arduino IDE will add new functions  for watchdog - but this code should not be affected - but may be the watchdog activation in variant.cpp

Wurstnase, 2015-06-15:
I've made the watchdog implementation also. With Arduino 1.6.5 we don't need to change the variant.cpp anymore. We just need to overwrite a weak function.

luc-github, 2015-06-15:
Ho - just see your implementation - sorry I missed it
I do not use 1.6.5 as it is not yet released but only in hourly builds 

So I close - issue thanks

closed 2015-06-15 by luc-github

Wurstnase, 2015-06-15:
I'm at 1.5.8 actually because this is only compatible with eclipse. But with 1.6.5 I will also go with it.