tonnico/Marlin4Due #35: Printer freeze when modify a flow during a print

giutrec, 2015-06-15:
I've looks a little bug when modify a flow variables in tune menu during a printing, the printer freeze all stepper. Maybe i know the solution i try it friday....

Wurstnase, 2015-06-15:
Does it freezes completely or just for ~2 minutes? What's about the heaters in this time?

Maybe it's an issue with the counter.

giutrec, 2015-06-16:
Its freeze after one o two minutes that the flow change was made and once there remains freezed until you restart a printer.
For the heaters i looks its friday and made a little report

Wurstnase, 2015-06-16:
Please try my experimental branch. I change some timing-parts. If you already use the actual master branch you can use the same config and config_adv. If you are using only DRV8825, A4988 or similar you also don't need to update your configs.

giutrec, 2015-06-26:
ok, now i try it and report you.... :) thanks

giutrec, 2015-07-04:
Hum, i've tryed more to made works a experimental version, but it's not work because have a more problem such as endstops, heaters, display.....

Sorry :(

Wurstnase, 2015-07-09:
You should replace the config-files from the experimental branch with your personal 'old' ones.

Wurstnase, 2015-07-28:
please reopen if this still happened.

closed 2015-07-28 by Wurstnase