tonnico/Marlin4Due #40: Auto Bed Leveling für Delta funktioniert nicht

chinchilla1982, 2015-09-08:
Hallo,

habe einen Delta mit dem Radds board und versuche das Auto Bed Leveling zum laufen zu bringen.
hab dabei folgendes Problem. Wenn ich den Befehl G29 ausführe fährt der Drucker richtigung Home Max und bleibt beim erreichen eines Endstops stehen. Im pronterface schreibt er dann auch Werte rein siehe Screenshot. Wenn ich dann den Drucker manuell steuere fährt er endlos nach oben bis ich den Reset Button drücke. 

hast du zufällig irgend eine Idee an was das liegen könnte?

drucker sollte eigentlich Richtung Heizbett fahren. 
der Endschlater ist auf Z-min eingesteckt
hab auch schon einige Konfigurationen durchprobiert.

danke im voraus

chinchilla1982, 2015-09-08:
[image]

Wurstnase, 2015-09-09:
Sorry. Ich habe keinen Delta und kann dir da auch nicht helfen. Vielleicht kann dir wer im reprap-Forum weiterhelfen.

Ich kann dir auch nicht sagen ob das Leveln aktuell überhaupt bei nem Delta komplett funktioniert. Da müsstest du dann ggf. auch mal bei MarlinFirmware nachfragen.

Im github ist allerdings Englisch die Hauptsprache. Auch wenn ein paar Leute deutsch können.

closed 2015-09-09 by Wurstnase