tonnico/Marlin4Due #41: REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER with RADDS

clefranc, 2015-10-12:
Hi,
I'm currently in the process of upgrading my delta from Rumba to RADDS. My delta already have a REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER, so I wish to use it with the DUE.

```cpp
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
```

I'm using the same pins as the RADDS_DISPLAY (using jumpers), added another `#elif`in `pins_RADDS.h` for the graphic LCD. No major code modification. It doesn't work.

I've switched the display for a REPRAP_DISCOUNT_SMART_CONTROLLER (same pinout, no code modification from above test), it works:

[image]

Before going deeper in the code, I want to know if it's already implemented. Can this full graphic LCD be used with Marlin4Due or it needs more code work?

Thank you
Christian

Wurstnase, 2015-10-12:
Yes, it will work. Someone get it to work with Repetier.
https://github.com/luc-github/DUE-RADDS-GLCD
But you can't use this pins. They are connected hard to a lcd encoder and can't be used for anything else. Bad design in that case.

clefranc, 2015-10-12:
Thanks, I'll report back if I can make it work.

AnHardt, 2015-10-12:
https://code.google.com/p/u8glib/wiki/device
tells about the 7920 is implemented for due.
Marlin supported the 7920 before u8glib did. For that we have still the extra driver in 
ultralcd_st7920_u8glib_rrd.h.
Maybe it's worth to try the (later) original u8glib driver. (u8g_dev_st7920_128x64_4x_sw_spi)

Wurstnase, 2015-10-13:
I'm already running a fullgraphic on my due. So generally it works.

Sage1978, 2015-10-19:
ultralcd_st7920_u8glib_rrd.h  Less line delay resulting CLK PIN not operate normally

static void ST7920_SWSPI_SND_8BIT(uint8_t val)
{
  uint8_t i;
  for( i=0; i<8; i++ )
  {
    WRITE(ST7920_CLK_PIN,0);
    delayMicroseconds(2);
    WRITE(ST7920_DAT_PIN,val&0x80); 
    val<<=1;
    delayMicroseconds(2);
    WRITE(ST7920_CLK_PIN,1);
    delayMicroseconds(2);//ADD THIS
  }
}

leseaw, 2015-10-19:
on the picture i cant see an full graphic :)
or i´m wrong? :)
its an 20x4

Wurstnase, 2015-10-19:
```
static void ST7920_SWSPI_SND_8BIT(uint8_t val)
{
uint8_t i;
for( i=0; i<8; i++ )
{
WRITE(ST7920_CLK_PIN,0);
delayMicroseconds(2);
WRITE(ST7920_DAT_PIN,val&0x80); 
val<<=1;
//delayMicroseconds(2); //REMOVE THIS
WRITE(ST7920_CLK_PIN,1);
delayMicroseconds(2);//ADD THIS
}
}
```

I guess, this is the correct solution.

clefranc, 2015-10-19:
@leseaw You're right it's a 20x4 I'm using to test the pins.

clefranc, 2015-10-19:
Thank you @Sage1978 & @Wurstnase :

[image]

Wurstnase, 2015-10-20:
@clefranc which pins do you use? Can you make a picture from it?

clefranc, 2015-10-20:
@Wurstnase The graphic display doesn't work anymore. It was working as you can see in the picture above, but I got garbage when moving wires, so I used the wiring described here: https://github.com/luc-github/DUE-RADDS-GLCD. I've remapped pins_RADDS.h, it doesn't even display the boot screen, only garbage updated each second.

I've rewired/remapped it back to the original working state, no luck. I've tested the character LCD again, it's fully functional.

Don't have much time to troubleshoot this week, will try another GLCD, reinstall the IDE...

Sage1978, 2015-10-21:
I use my own modified version produced FD V2We have encountered problems after the startup screen display stuck The reason is found in the ADC program FSR_SENSOR uninitialized not be started I will amend this after the normal, but the temperature sensor is not installed there will be similar problems

Wurstnase, 2015-10-21:
On my first try with a FullGraphic I've got some issues with the display. I used then the standard implementation from u8glib instead of rrd. This is much slower but works. After that I went back to rrd and magically it works again.

Wurstnase, 2015-10-21:
@Sage1978 I found the FSR-things. That was an old test and I forgot this part. I will delete all fsr-things later.

closed 2016-02-29 by Wurstnase