tonnico/Marlin4Due #42: Marlin paused the print for a few seconds

svenp, 2015-10-16:
Marlin is pausing the print for a few seconds after 50 -60 minutes of printing time.
Then you can find in the serial window there is a "k" instead of "ok" 
Marlin is connected to Windows7 with 250000 Baud at programming port.
With marlin 1.02 at Mega2560 there was no Problem with 250000 at Win7.

[image]

Wurstnase, 2015-10-16:
When you received just a 'k' your cable has an issue. Maybe too long? But it's not an issue from Marlin or Marlin4Due.

I used 250000 baud some time ago, but now I'm on Linux and there is 115200 more stable or common.

svenp, 2015-10-16:
The cable is new and only 15 or 20cm long, but you are right this could be the problem.
When the print is finished in some hours, i would also try 115200. 
Is it better to use the programming port or the native usb port?

Wurstnase, 2015-10-16:
Never tried the native port and this should need some modifications in the code.

If you have some issues with a stable connection AND using Repetier Host you could try ```#define NO_TIMEOUTS 1000```.

Your print only go further because Repetier asks every 2 or 3 seconds for the temperature. And then you get the next 'ok'.

svenp, 2015-10-18:
With 115200 same cable there is no problem. Everithing is working.

closed 2015-10-19 by Wurstnase