tonnico/Marlin4Due #44: How to set up micro-step stepping motor using 1/128 and 1/256

Sage1978, 2015-10-24:
I use DRV8711 stepper motor driver IC, and is set to 1/128.
But no matter how marlin motor movement distance is not properly set, it looks like 1/32.

Wurstnase, 2015-10-25:
Normal driver are set up with jumpers. I don't know this drivers.

Sage1978, 2015-10-26:
Whether this version of the firmware can support 128 or 256 micro-step micro step

driver IC is set to 1/128
Firmware set to 1/128
#define DEFAULT_AXIS_STEPS_PER_UNIT {640, 640, 640, 190}
Travels the same distance between the test results and 1/32
#define DEFAULT_AXIS_STEPS_PER_UNIT {160, 160, 160, 190}
This is wrong, the line should be different close
The same hardware set, Repetier firmware is OK

Wurstnase, 2015-10-26:
Don't know what your problem is exactly.
When I use 1/16 stepper driver I set up my system with 80 steps/mm. With 1/32 I need 160 steps/mm. So I have no idea where your problem should be. It is not Marlin...

Wurstnase, 2015-10-26:
Ah, I have an idea of your issue.
You should overwrite your eeprom.
Load your values from the config file with M502 and save them with M500.

Sage1978, 2015-10-27:
Thanks for your help
EEPROM does not update in question
Currently 1/256 microstepping smooth operation

Wurstnase, 2015-10-27:
Can we close this issue?

closed 2015-10-27 by Wurstnase