tonnico/Marlin4Due #45: TEMP_SENSOR 1 ~ 3 can not be used

Sage1978, 2015-11-05:
To use TEMP SENSOR do not use environmental monitoring found
#define TEMP_SENSOR_0 1
#define TEMP_SENSOR_1 1
#define TEMP_SENSOR_2 1
#define TEMP_SENSOR_3 0
#define TEMP_SENSOR_BED 1

Program miss more than one;
    #if HAS_TEMP_0
      START_TEMP(0);
    #endif
    #if HAS_TEMP_BED
      START_BED_TEMP();
    #endif
    #if HAS_TEMP_1
      START_TEMP(1)//miss ;
    #endif
    #if HAS_TEMP_2
      START_TEMP(2)// miss;
    #endif
    #if HAS_TEMP_3
      START_TEMP(3)//miss;
    #endif

There are other places also have changed but I do not know will not be affected

Sage1978, 2015-11-05:
temperature.cpp  
case PrepareTemp_1:
      #if HAS_TEMP_1
        START_TEMP(1)//miss ;
      #endif
      lcd_buttons_update();
      temp_state = MeasureTemp_1;
      break;
    case MeasureTemp_1:
      #if HAS_TEMP_1
        READ_TEMP(1);
      #endif
      temp_state = PrepareTemp_2;
      break;

    case PrepareTemp_2:
      #if HAS_TEMP_2
        START_TEMP(2)//miss ;
      #endif
      lcd_buttons_update();
      temp_state = MeasureTemp_2;
      break;
    case MeasureTemp_2:
      #if HAS_TEMP_2
        READ_TEMP(2);
      #endif
      temp_state = PrepareTemp_3;
      break;

    case PrepareTemp_3:
      #if HAS_TEMP_3
        START_TEMP(3)//miss ;
      #endif
      lcd_buttons_update();
      temp_state = MeasureTemp_3;
      break;
    case MeasureTemp_3:
      #if HAS_TEMP_3
        READ_TEMP(3)//miss ;
      #endif
      temp_state = Prepare_FILWIDTH;
      break;

Wurstnase, 2015-11-06:
Please make a pull request, so I can merge it.

Wurstnase, 2015-12-03:
fixed

closed 2015-12-03 by Wurstnase