tonnico/Marlin4Due #48: Marlin4Due for Duet 0.8.5?

rsatue, 2015-11-18:
Hi!

Do you think Marlin4Due could run on a Duet 0.8.5? 

According to their specs, the Duet is compatible with the Arduino Due and it combines the Arduino Due microcontroller with 5 stepper motor controllers (v0.8.5), Ethernet, Hi-Speed SD card slot and more. 

What should be worked out to make it run? The pins.h file?

Thanks in advance.

Wurstnase, 2015-11-18:
Could run. 
You need a new pins.h yes. But Ethernet isn't supported. SD-Card uses SPI afaik.
I have no DUET. But there is also no support for the control of the stepper-current. Just one thing I know what needs some additional work.

rsatue, 2015-11-18:
Thanks for the quick reply!!

We have a Duet and time to work on it. If you could guide us may be we could work on it using some parts of the original's Duet firmware. 

We could live without Ethernet. And probably also without the SD. Our idea is to connect a tablet with the USB and load the gcode file on the tablet.

Wurstnase, 2015-11-18:
What is wrong with Reprap-Firmware from @dc42?

rsatue, 2015-11-18:
We have a cartesian 3D printer with two independent carriage extruders along the x-axis and this is not supported.

Wurstnase, 2015-11-18:
I would guess that this could be done with some special gcodes?
Afaik, Reprap-Firmware has a couple of this. For homing and maybe also for switching between hotends? There you could drive the actual hotend to the save spot, change the pins, and use the other hotend?
I guess changing reprap-firmware is much easier and faster (and you loose no features) than rewriting much more code in Marlin.

rsatue, 2015-11-18:
I put the question yesterday at their github and this is what Ian from the tech support said:

As far as I'm aware, at the moment there is no functionality to define a particular motor axis, eg X1 or X2, to a tool, apart from defining which extruder drive is used. Probably, the functionality of the tool definition gcode M563 ( http://reprap.org/wiki/G-code#M563:_Define_or_remove_a_tool ) would need to be extended to be able to set an axis to use. How to handle the tool change, and coordinating the coordinates of each separate X axis, would need to be worked out too. Unfortunately, we have no plans to add this functionality at the moment, but that doesn't mean someone may not take up the challenge to add it!

That's why I am looking for alternatives.

closed 2015-12-03 by rsatue