tonnico/Marlin4Due #49: Add travis-ci for testing several modules / arduino versions / conditions

luc-github, 2015-11-20:
Hi not sure you need it, but I did this on my git-hub and you may be interested as you also work on Due.
Current script test only IDE 1.6.5 and Due 1.6.4 but adding version 1.6.6 for both is easy - I plan to do later, because I did not found yet the way to display different status according IDE/Module version, may be using a separate branch ?

This also help me to test all possible configurations by automating the build tests, may be you will find it useful.

Here is my script FYR : https://github.com/luc-github/Repetier-Firmware-0.92/blob/master/.travis.yml

Wurstnase, 2015-11-20:
Thanks for sharing it.

Acutally I have no plan to move much more ahead with Marlin4Due. I'm waiting for the RC to release on MarlinFirmware and will patch this then. If there are some Bugs, I try to fix them. But I won't go with Marlin than anymore.

The library change is way too much work for me. Maybe some others want to take this work.

luc-github, 2015-11-20:
Understood, thanks for your work ^_^

luc-github, 2015-11-24:
sorry forgot to close it

closed 2015-11-24 by luc-github