tonnico/Marlin4Due #50: question

ihouses, 2015-12-16:
Hello!

How is the status of this project? 

Recently I bought a delta printer and start using Marlin and I figured out that Arduino Mega is not enought for that task. I have some experience programming and not too much with 3d printers. I am using the typical setup with Arudino Mega and Ramp1.4.

dmason1992, 2015-12-16:
Hi Ihouses,

I am not certain of how it is going but I am in a similar position in that I am looking at using Ramps-FD along with an arduino due for my new printer i'm building. Looking at the activity there has been a fairly recent commit but overall not allot since summer this year. 

I'm happy to start getting involved in helping fix bugs etc but am also not the best programmer in the world. My final goal is to have a fast moving coreXY printer which has closed loop feedback on the x&y axis :)

ihouses, 2015-12-16:
Hello again,

I was looking at http://smoothieware.org/smoothieboard, it is a interesting board as well and the firmware looks much mature than others based on Arduino DUE.

dmason1992, 2015-12-16:
This is potentially true, although repetier host supports due, but smoothieboard doesn't give the same level of modification possibilities and also costs quite a bit more

zone01-org, 2016-01-28:
Take a look here http://www.instructables.com/id/Vulcanus-V2-Reprap-3D-Printer/ to see a practical aplication.

Wurstnase, 2016-01-29:
I'm using this Firmware now for a year. It's close to the AVR-Version. But actually I don't have time for maintenance.

closed 2016-02-29 by Wurstnase