tonnico/Marlin4Due #51: Marlin4Due for laser - plasma cnc

mkeyno, 2016-02-19:
hi guys , does any one know any effort to applying this frimeware for 2d cnc such as  laser and plasma,?
I used Marlin modified by cool guy https://github.com/TurnkeyTyranny/buildlog-lasercutter-marlin run on the RAMPS1.4 , it work partially well but the guy not support it anymore and suffer for the low pulse rate , I think the Due base would be promising for such printer 
any help really appreciated

HakanBastedt, 2016-02-26:
Some time ago I took the changes from buildlog-lasercutter-marlin and applied them to the current RC of Marlin. The changes were well encapsulated with #ifdef LASER and should be straightforward to apply also to the code here. There is really only one CPU-dependent part and that is the PWM for the laser pulse and for that one Timer and a PWM-pin are needed. Without knowing in detail I just assume that the Due has a free Timer; that code is in laser.cpp and laser.h.

mkeyno, 2016-02-29:
tnx @HakanBastedt  however i've decided to test my cnc with this due and firmware and publish it here

HakanBastedt, 2016-03-01:
Sure, I didn't mean you should use my work. But the work needed is the same for due.
I am building a CO2 laser now and would like to have faster step rate than what the Mega can deliver and have started to look into what is needed to switch to Due Marlin.

As far as I can see the only challenge is to set the intensity of the laser with pwm. One _could_ use analogWrite for the pwm, however the pwm frequency is a bit low around 500 Hz. It needs to be up to 20 kHz. In HAL.cpp there are the tone() and servo-pwm functions (don't remember the name) that can be used. Pwm on due is a bit different from the Mega. I think the right thing to do is to use one of the PWM generators in the Due to generate PWM on one of the the servo pins (pwm capable).  https://github.com/antodom/pwm_lib is a lib I have glanced at to do exactly this.

When this is done, it is "only" to copy in the other code changes into Marlin_main,cpp, planner.cpp and stepper.cpp.

mkeyno, 2016-03-01:
thanks @HakanBastedt  for reply and it seems this library has promising feature and I hope finally you could enrich the code for multi CNC , however I have plan to test all Due support printer source code for my laser CNC, 
Actually I ran my laser machine with RAMPS1.4 with modified marlin version  https://github.com/TurnkeyTyranny by cool guy , this guy use special header file to enable and  control intensity of laser, however he is not willing to continue his work and the code so old and suffer from minor defect such as speed control & run-time execution, for example if you decide  stop the streaming  feed , it will continue  and execute the what is in its buffer whereas in GRBL you could easily halt the machine anyway , also I've installed the HC-06 Bluetooth module and wirelessly control the machine and that's why I'm going to find best code to have support the elegant ESP8206 module 
by the way have you check the other codes

HakanBastedt, 2016-03-02:
I integrated the laser raster functionality from https://github.com/TurnkeyTyranny in my copy of Marlin4Due yesterday. It compiles, but I need to add DUAL_Y_STEPPERS and endstops before testing. I took the pwm code from https://github.com/cloud-rocket/DuePWM, quite straight forward.
If you don't need raster functionality you can just use G1 and M42 Px to switch on/off the laser. You can even do raster that way but the amount of streaming G-code will slow down. If you have only X & Y then Z needs to be disabled in some way. This is how I first ran my Core-XY laser-diode cutter. 

I am not sure what you mean with speed control problems in TurnkeyTyranny's code, can you elaborate on this? What I can see is that the G7 raster engraving code takes acceleration and deceleration into account, while e.g. lasersaur's G8 code adds acceleration and deceleration distances outside of the raster area. I guess it would be possible to add G8 as well.

Soft emergency stop/pause is interesting. There is the M112 command, that reacts as soon as Marlin sees M112 in the stream, but it kills everything and waits for reset. Switch of the laser, stop movements and pause is interesting, I guess it can be done. Perhaps operated by a switch connected to a pin as well. Seems useful for a laser, maybe less useful for a 3D printer.

mkeyno, 2016-03-02:
hi @HakanBastedt , I think many code here suffer from standardization , whereas most common CAM software generate similar G-code, for example for laser cutting machine but it seems most  code here follow its own way 
 when I use TurnkeyTyranny's code F command not work during the print , I've try to modified inkspace plugin to store in EEPROM new speed before each line but its partially work and sometime the result is beyond expectation 
what I'm referring to GRBL  run-time command, means the code check emergency break in serial command not look into the general standard G-code command such as M112 and when it find the signal , cut down the feed whereas in Marlin TurnkeyTyranny's code it execute the last line in buffer before face the M112  code  and that's make big problem for CNC such as laser and plasma
I think ESP is good solution for that, because you can transmit the pin status along with serial buffer which is status of pin can check in loop for emergency abort or stop

HakanBastedt, 2016-03-04:
Hi, I looked at some other codes, and it seems most if not all use G1 Xx Yy Ss for laser cutting, and also raster engraving. Afai can see also Picengraver Lite does it this way. Smoothie only has G8 raster format as experimental from what I can see, normally engraving is there done with G1 Ss. I think now that TurnkeyTyranny's work stands out by not only having a G7 raster format that handles acceleration and deceleration, but also a PULSED mode which should also handle speed changes during the move. The CONTINUOUS mode will not do that, if the steppers slow down for some reason there will be more light. And TunrkeyTyranny is unique with the G7 raster format; the only G7 code generator I know o is the Inkscape plugin which works quite ok for me.

I tested Mega2560 Marlin with G1 and G7 raster encoding of a photo. With G1 I could reach something like 700 mm/min, while with G7 3000 mm/min was possible. I could be wrong exact figure but G7 was significantly faster anyway. Maybe I can reach 10000 mm/min with the Due?

Your problem with the F can be either with TurnkeyTyranny's old Marlin version or perhaps you reach the upper limit of which speed can be achieved. I only ran his version of Marlin for a short time so maybe I didn't notice it. I soon ported the laser code to current Marlin (I needed coreXY). And there was no such problems with the speed, it travelled with the speed I set if it to. There is an upper limit to how fast it can go, and processing a lot of small small movement takes some cpu time. That's why I am looking at the Due.
As you say, M112 is processed sequentially in TurnkeyTyranny's Marlin version. But in the current version of both Mega Marlin and Due Marlin M112 is processed as soon as it is seen in the input stream, so no waiting to empty buffers.

mkeyno, 2016-03-04:
I don't know caz the code has not been modified for more that 2 years but his way to build special header file to control laser is elegant , also I believe his inkspace plugin very un mature compare to other laser plugin 
have you check this code https://github.com/ambrop72/aprinter , I've talk with Ambroz and he put some good modification in his code such as pin status change , laser Active pin and I suggest  have look in his codes 
thanks @HakanBastedt

HakanBastedt, 2016-03-06:
I looked briefly at aprinter and to me it is like all the others. He does take speed change into consideration with a pulsed pwm, where laser intensity changes at intervals in time. Tyranny changes intensity at intervals in distance. About the same I think. And there is no real raster engraver, only with small g1's.

Anyway, I am now ready with the integration of TurnkeyTyranny's laser code in my Marlin4Due, it is in the Laser branch if anyone is interested. With the G7 raster engraving command I can engrave at 20000 mm/min, without the reprap full graphic display engraving can be done at 30000 mm/min. So I am happy with the speed, that should be sufficient. Engraving was with 270 dpi, and 100 steps/mm for motors/drivers, the distance was like 300 mm in X so there was time to get up to full speed. It is a tremendous improvement in speed compared to the Mega 2560.

mkeyno, 2016-03-06:
hi @HakanBastedt , is this your finial decision to  use TurnkeyTyranny's laser code in  Marlin4Due ? does your CNC have peripheral device such as seneloed valve  ?I cant find laser configuration in your code , can you let know which header file is  ? also  I'm about to finalize my Due  shield PCB and I let you know buddy, but right now my focus is to embedded ESP Link in it

HakanBastedt, 2016-03-07:
Yes, at least for the moment. I am used to Marlin and know the code and that is a big benefit for me. And I like the G7. If it breaks I am able to fix it. Best for me, not best for everyone. And it is done already, the code is in https://github.com/HakanBastedt/Marlin4Due/tree/Laser, I have tried to encapsulate it with #if LASER but I sure have missed things. Code changes are all over the place (almost). I made a short video showing the engraving performance: https://www.youtube.com/watch?v=aUFzQa_7U6Y Still need to lign up the mirrors and do some actual cutting.

At this point I have no peripherals, but the RAMPS-FD card has six (four high-power) 12V outputs driven by MOSFETs, so at least for the cooling pump/fan it should work.

Wurstnase, 2016-03-08:
You could ask or search @VDX in the reprap-forum. He uses Marlin4Due for laser-applications for some month.

mkeyno, 2016-03-09:
thanks @HakanBastedt and wlecome @Wurstnase  to join this talk, actually I read your code as well as  Mralin due code of guys like @[JRodrigoTech ](https://github.com/JRodrigoTech/Marlin-DUE) , @[MagoKimbra](https://github.com/MagoKimbra/MarlinKimbra4due), @[ambrop72](https://github.com/ambrop72/aprinter), @[Luc](https://github.com/luc-github/Marlin4Due), however  I believe all those code suffer form lack of standards for CNC cutting machine such as Plasma and Laser and it is not surprising whereas Marline originality created for 3D printer , many Marline potential used for extruder and temperature controlling,  whereas we don't need it in cutting machine and instead need used Z-probe or Auto bed leveling feature  to constantly check the distance between martial and nozzle, I've tried to put function in loop to check the this prop pin status but I failed , maybe its because , I dont have enough knowledge to mess with interrupts  
I'm gonna consult with all these guys here, maybe we could then created the standard version Marlin for the cutting machine

luc-github, 2016-03-09:
Just for reference repetier fw is supposed to support some laser and cnc feature now, you may want to have a look at what it is doing  (https://github.com/repetier/Repetier-Firmware/blob/work092/src/ArduinoDUE/Repetier/Configuration.h#L702-L729).
Sorry I cannot say more on it as I do not have cnc milling neither laser machine so not I am not sure what are the full needs, but I plan to build both soon (still looking for info how to build them, so once setup, I will be able to help more)

closed 2016-03-29 by Wurstnase