tonnico/Marlin4Due #52: Servo pin for ramps-fd same as sdss pin

DenniZr, 2016-02-23:
For the ramps fd the servo3 (the fourth one) is pin D4. Servo0 is D11. Sdss is also D4. Why are the sdss the same as the servo pin. When u read http://www.geeetech.com/wiki/index.php/File:RAMPS-FD-Schematic.pdf the servo03 is said to be D3. Why does the marlin pin say D4?

Wurstnase, 2016-02-24:
No reason. I never checked this.

DenniZr, 2016-02-24:
I want to use the servo03 not as a servo, but as a data communication line. Or actually, i am allready. but i get strange signals on this line. Is the servo pin really D4? If you can tell me that.
Geeetech says it's D3.

Wurstnase, 2016-02-24:
Can't say it. I don't have any schematics. Just set it to -1 or comment it out and try. That's how I would do it.

closed 2016-03-29 by Wurstnase